DhrBaksteen/ArduinoOPL2 #9: Type error in opl.py of serialIface example

DhrBaksteen, 2017-08-01:
This issue was reported on the Tindie page:

> Play "sonic.vgm" on COM5
(Ctrl-C to stop)
Traceback (most recent call last):
File "play.py", line 74, in 
handle_arguments()
File "play.py", line 65, in handle_arguments
device = opl.ArduinoOpl(portname)
File "C:\Users\Fu▀pilz Stinkstiefel\Desktop\ArduinoOPL2-master\ArduinoOPL2-master\examples\SerialIface\opl.py", line 28, in init
self.status(self.READYCMD)
File "C:\Users\Fu▀pilz Stinkstiefel\Desktop\ArduinoOPL2-master\ArduinoOPL2-master\examples\SerialIface\opl.py", line 63, in _status
tx_txt = 'Tx: %s' % ' '.join('%02x' % b for b in last_tx)
File "C:\Users\Fu▀pilz Stinkstiefel\Desktop\ArduinoOPL2-master\ArduinoOPL2-master\examples\SerialIface\opl.py", line 63, in 
tx_txt = 'Tx: %s' % ' '.join('%02x' % b for b in last_tx)
TypeError: %x format: a number is required, not str

bsutherland, 2017-08-03:
Hi,
Tried to reproduce with other vgm files, but couldn't. Could you upload sonic.vgm?
Are they using Python 2 or 3? Pretty sure the script requires Python 3.

pioneerman2k1, 2017-08-24:
Hi bsutherland,

you are tataly right. i was using Python 2. Meanwhile i figuerd out by myself. Thanks a lot.

Btw. is it possible zo implent more filetypes for playback?

bsutherland, 2017-08-24:
Glad it's working. No plans to implement support for other file formats at this point, but what did you have in mind?

pioneerman2k1, 2017-08-24:
Well, there is code for "PlayDRO" and "PlayIMF" only. The standart Soundformat for the YM3812 is VGM/VGZ this would be nice to create an "PlayVGM(Z)"  *smile*
The problem is the timing when converting other Filetypes to DRO or IMF. Sometimes the timing is correct but sounds awful..

( http://vgmrips.net/packs/chip/ym3812 )

bsutherland, 2017-08-25:
I only wrote the SerialIface code. If you load that onto your board, the Python script play.py supports both vgm and vgz.
https://github.com/DhrBaksteen/ArduinoOPL2/tree/master/examples/SerialIface

closed 2018-01-20 by DhrBaksteen