DhrBaksteen/ArduinoOPL2 #11: SPI device error

toxicsting, 2018-02-10:
using raspberry pi 3

Im getting an error when I enter this command.
`
pi@raspberrypi:~/ArduinoOPL2/examples_pi/opl2play $ sudo ./opl2play tunes/duke2.imf
Unable to open SPI device: No such file or directory
pi@raspberrypi:~/ArduinoOPL2/examples_pi/opl2play $ pwd
/home/pi/ArduinoOPL2/examples_pi/opl2play
`
what am I doing wrong?

DhrBaksteen, 2018-02-11:
Can you check if SPI is enabled in your /boot/config.txt? Look for 'spi=on'. There should be a line like 'dtparam=spi=on' in that file.

toxicsting, 2018-02-12:
well the SPI setting was committed out so i re-enabled it but now the audio comes in as a series of pops. in case its  a wireing issue here are some pics of the wireing. 
[image]

[image]
[image]

DhrBaksteen, 2018-02-12:
Thanks, the pictures are a real help. First thing I notice is that the 74595 (IC6) is mounted in the wrong direction. Please flip it around. 

If that doesn't help then we need to see if the wiring of your breakout board may be different.

toxicsting, 2018-02-24:
sorry its taken me so long to get back to you.

this is what i got from running this code.
` sudo ./drums`

[soundblaster .zip](https://github.com/DhrBaksteen/ArduinoOPL2/files/1753675/soundblaster.zip)
[image]
note: i put a set of headphone speakers at point blank to my mic to capture this (not earbuds).

DhrBaksteen, 2018-02-25:
Alright, let's do some further digging then...

Do you get this clicking with all of the examples or just the drums one?

I notice that your breakout board has separate MOSI and SCLK pins labeled. What happens if you connect DATA to MOSI and SHIFT to SCLK? Or when you connect the OPL2 board directly to the Pi?

Can you run `gpio readall`? You should see that physical pins 12, 16 and 18 (middle column) are labeled 1, 4 and 5 in the wPi column and pins 19 and 23 are labeled MOSI and SCLK. If this is different for you then try connecting the board like this:
A0 - wPi 4
DATA - MOSI
LATCH - wPi 1
SHIFT - SCLK
RESET - wPi 5

If all of the above fails then let's see if the OPL2 chips is doing something. You can check by probing with a little speaker between pin 4 of IC2 and GND.  You should hear hissing / noise while you run the drums example.

DhrBaksteen, 2018-02-25:
I forgot to mention, since the 74595 was connected the wrong way round before, this chips will get very hot in that way. It will not instantly kill the chip, but leaving the OPL2 board connected like this for a long time will not do it any good. If you have a spare 74595 lying around then you may want to swap it to make sure it didn't get damaged.

toxicsting, 2018-03-03:
here is what ive done so far
I notice that your breakout board has separate MOSI and SCLK pins labeled. What happens if you connect 

DATA to MOSI and SHIFT to SCLK? _____nothing diferent happend execpt i noticed more static(might be my 

imagination)

gpio

[image]

nothing is wrong that i can tell

checcked headphone jack continuity ---looked good

conect to raspberry pi directly?_____static but only when trying to play a file

[image]
[image]

noticed that while adjusting R5 the headphones gave a loud hum is that normal?

I dont have any spare chips.

Thank you for the help you have given me so far!!

DhrBaksteen, 2018-03-04:
There shouldn't be any humming while adjusting the volume.

It's starting to look like a component got damaged. We can try pinpointing it by testing the following while running the demotune example. Some of these signals will be difficult to hear unless you can probe using an external speaker with a little amplifier.

Pin 1-7 of IC6 - You should hear loud clicking noises 
Pin 3 of IC2 - Hissing
Pin 4 of IC2 - You should hear the tune playing loud and very distorted
Pin 4 of IC3 - Tune is playing on low volume, slightly distorted
Pin 1-2 of IC3 - Tune is playing clearly, slightly higher volume
Pin 3 of IC4 - Tune is playing clearly at low volume
Pin 5 of IC4 - Tune is playing clearly at adjustable volume usinf R5

DhrBaksteen, 2018-04-07:
Closing the issue due to inactivity

closed 2018-04-07 by DhrBaksteen