DhrBaksteen/ArduinoOPL2 #12: terrible noise

romeurolo, 2018-02-14:
hi mate, i receive the board and connected to my Chinese Arduino NANO clone, and Nothing.
Some beers after i found that the problem is the wires not connected in the correct place.
Now the board it's working but you can only hear some sound in the middle of the noise.
My questions are: Chinese arduino sucks? My computer power switch sucks? I try to power up the board whit an external USB power switch, but the noise still remains.
I will post some photos and audio file to show what im talk about.
Thank for the Board, Romeu

romeurolo, 2018-02-14:
[image]
[image]
[image]
[image]

romeurolo, 2018-02-14:
https://youtu.be/Ku73E8Qbxf0

DhrBaksteen, 2018-02-14:
Hmm this puzzles me a bit.. I've heard it produce some strange noises while developing the board but not like this.

I take it that the pins on your breadboard with the Arduino are making a proper connection. 

Let's see if we can diagnose it... Can you probe the board with an amplified speaker or headphones between a pin and GND while the demo tune sketch is running? You should get the following when probing:
Pin 4 of IC2 - You should hear a distorted version of the tune
Pin 2 of IC2 - You should hear the tune on low volume, less distorted
Pin 1 of IC3 - You should hear the tune clearly but not very loud
Pin 5 of IC4 - You should hear the tune regularly and volume can be adjusted with the pot

Another thing to try is what happens when you sweep the volume carefully? If there is a very narrow setting where the noise goes away and you can hear the music softly then it could be that you have a broken capacitor.

romeurolo, 2018-02-14:
I try and the result are:
Pin 4 of IC2 - Distorted version of the tune
Pin 2 of IC2 - I can't hear nothing
Pin 1 of IC3 - I hear the tune on low volume, less distorted
Pin 5 of IC4 - I hear the tune low whit background noise

In the pot i try several times what you told me to do but the noise still remains.

DhrBaksteen, 2018-02-17:
I'm trying a few things here, but so far it looks like a damaged amplifier

DhrBaksteen, 2018-02-18:
Can you please measure if you have a short between pins 5 and 6 of IC4?

romeurolo, 2018-03-09:
Hi, change all the Ic's, i made a nice 5 volts power supply and the noise remains. And that left me whit 2 problems, some passive component damage or the Chinese Duino sucks....

cbmeeks, 2018-03-09:
Do you have a genuine Nano/UNO to test?

romeurolo, 2018-03-09:
Not at this moment, but i have order some arduinos. I have to wait until they arrive.

DhrBaksteen, 2018-03-09:
I would be surprised if it's due to the Arduino. The other circuitry on your board is not interfering with het Arduino?

vincentbernat, 2018-04-07:
@romeurolo I have been used an Arduino Nano similar to yours without any issue (and with no additional components, just powered by USB).

DhrBaksteen, 2018-05-21:
Closing this issue due to inactivity.

closed 2018-05-21 by DhrBaksteen

ashermja, 2020-03-06:
Hi @DhrBaksteen, love the board but I'm having some problems with a large amount interference. I have a pre made board. I think it could be a capacitor issue as there is a narrow volume setting where most of the noise goes away.

DhrBaksteen, 2020-03-07:
Can you provide a sample of the noise? Usually that diagnoses the problem quite quickly. One possible cause is that the power supply introduces unwanted noise.

ashermja, 2020-03-08:
Thanks for getting back to me. I've tried quite a few different power supplies to see if it makes any difference, but all were the same.

https://www.dropbox.com/l/scl/AABzNk0U9ULBcg5PE0voerGhlTv5gyqmbpc
https://www.dropbox.com/l/scl/AADj3ODCuLbxYI0jT9jStafam5K0Bt1fHm4

DhrBaksteen, 2020-03-09:
I wasn't able to access the files unfortunately. They were private to you.

In the mean time to rule out the LM386 you can try by removing the chip from its socket hooking up some externally amplified speakers between the pins 3 and 4.

DhrBaksteen, 2020-03-09:
Thanks for getting me access. It sounds as if the filter is having some issues and likely this is indeed one of the caps. Can you try the above and let me know? I will get back to you later today

ashermja, 2020-03-09:
Just tried removing the LM386 same and testing pins 3 and 4, same interference.

DhrBaksteen, 2020-03-09:
Ok, then I suspect that C1 is causing some issues.

Can you check one last thing? Try to reseat the ICs 2 and 3. And can you check that on the bottom of your board non of the pins are bent so that they are touching?

ashermja, 2020-03-11:
Tried reseating IC 2 and 3 and still the same problem. The bottom of the board looks fine, nothing is is touching or bent. So I guess it's C1?

DhrBaksteen, 2020-03-11:
That is probably this issue indeed. Can you send me a message on Tindie? Then let's get this resolved! :)

ashermja, 2020-03-11:
Great, thanks