DhrBaksteen/ArduinoOPL2 #27: Teensy 2.0++ Midi Support

pioneerman2k1, 2018-06-25:
Is it possible to add support for the teensy 2.0++?
The Teensy is able to act like an USB Midi device.

DhrBaksteen, 2018-06-25:
Yes that is possible. In fact I have a Teensy ordered that I hope to receive by the end of this week to start implementing this :)

pioneerman2k1, 2018-06-25:
Nice! In that case u can use it (like MUNT) as an Midi Device. So u would be able to Play on an MIDI-Keyboard and use the OPL2 Board as Output.
Cant wait for it!

saschaamend, 2018-06-27:
Sounds nice!
Another thing is, that u can use it as an Midi-Device in DosBox as Output Device. That would be so nice!

saschaamend, 2018-07-03:
There was missing the usb_names.h in the lib.

Got this Error:

**_Arduino: 1.8.5 (Windows 10), TD: 1.42, Board: "Teensy++ 2.0, MIDI, 16 MHz, US English"

In file included from C:\Users\user\AppData\Local\Temp\arduino_build_815943\sketch\OPL2.cpp:49:0:

C:\Users\user\AppData\Local\Temp\arduino_build_815943\sketch\OPL2.h:19:0: warning: "ARDUINO" redefined

  #define ARDUINO      0

 ^

<command-line>:0:0: note: this is the location of the previous definition

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Stream.h:24:0,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/../usb_midi/usb_api.h:6,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/usb_api.h:10,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/WProgram.h:22,

                 from C:\Users\user\AppData\Local\Temp\arduino_build_815943/pch/Arduino.h:1,

                 from C:\Users\user\AppData\Local\Temp\arduino_build_815943\sketch\OPL2.h:76,

                 from C:\Users\user\AppData\Local\Temp\arduino_build_815943\sketch\OPL2.cpp:49:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:71:21: error: 'Printable' does not name a type

  size_t print(const Printable &obj)  { return obj.printTo(*this); }

                     ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:91:23: error: 'Printable' does not name a type

  size_t println(const Printable &obj)  { return obj.printTo(*this) + println(); }

                       ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h: In member function 'size_t Print::print(int, int)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:65:65: error: call of overloaded 'print(int&)' is ambiguous

  size_t print(int n, int base)   { return (base == 10) ? print(n) : printNumber(n, 0, base); }

                                                                 ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:65:65: note: candidates are:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:53:9: note: size_t Print::print(const String&)

  size_t print(const String &s);

         ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:54:9: note: size_t Print::print(char)

  size_t print(char c)    { return write((uint8_t)c); }

         ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:55:9: note: size_t Print::print(const char*) <near match>

  size_t print(const char s[])   { return write(s); }

         ^

[... 196 lines cut ...]
  virtual void write(uint8_t);

               ^

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Stream.h:24:0,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/../usb_midi/usb_api.h:6,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/usb_api.h:10,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/WProgram.h:22,

                 from C:\Users\user\AppData\Local\Temp\arduino_build_815943/pch/Arduino.h:1,

                 from C:\Users\user\AppData\Local\Temp\arduino_build_815943\sketch\OPL2.h:76,

                 from C:\Users\user\AppData\Local\Temp\arduino_build_815943\sketch\OPL2.cpp:49:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:47:17: error:   overriding 'virtual size_t Print::write(uint8_t)'

  virtual size_t write(uint8_t b) = 0;

                 ^

Fehler beim Kompilieren für das Board Teensy++ 2.0._**

pioneerman2k1, 2018-07-03:
i got the same Error Message. Is it because of the usb_names.h?

pioneerman2k1, 2018-07-03:
Arduino: 1.8.5 (Windows 10), TD: 1.42, Board: "Teensy++ 2.0, MIDI, 16 MHz, US English"

In file included from C:\Users\Pione\OneDrive\Desktop\ArduinoOPL2-teensy_midi\examples\TeensyMidi\TeensyMidi.ino:19:0:

C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-teensy_midi\src/OPL2.h:19:0: warning: "ARDUINO" redefined

  #define ARDUINO      0

 ^

<command-line>:0:0: note: this is the location of the previous definition

In file included from C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-teensy_midi\src\OPL2.cpp:49:0:

C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-teensy_midi\src\OPL2.h:19:0: warning: "ARDUINO" redefined

  #define ARDUINO      0

 ^

<command-line>:0:0: note: this is the location of the previous definition

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Stream.h:24:0,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/../usb_midi/usb_api.h:6,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/usb_api.h:10,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/WProgram.h:22,

                 from C:\Users\Pione\AppData\Local\Temp\arduino_build_181674/pch/Arduino.h:1,

                 from C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-teensy_midi\src\OPL2.h:76,

                 from C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-teensy_midi\src\OPL2.cpp:49:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:71:21: error: 'Printable' does not name a type

  size_t print(const Printable &obj)  { return obj.printTo(*this); }

                     ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:91:23: error: 'Printable' does not name a type

  size_t println(const Printable &obj)  { return obj.printTo(*this) + println(); }

                       ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h: In member function 'size_t Print::print(int, int)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:65:65: error: call of overloaded 'print(int&)' is ambiguous

  size_t print(int n, int base)   { return (base == 10) ? print(n) : printNumber(n, 0, base); }

                                                                 ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:65:65: note: candidates are:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:53:9: note: size_t Print::print(const String&)

  size_t print(const String &s);

         ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:54:9: note: size_t Print::print(char)

  size_t print(char c)    { return write((uint8_t)c); }

         ^

[... 209 lines cut ...]

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/usb_api.h:10,

                 from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/WProgram.h:22,

                 from C:\Users\Pione\AppData\Local\Temp\arduino_build_181674/pch/Arduino.h:1,

                 from C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-teensy_midi\src\OPL2.h:76,

                 from C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-teensy_midi\src\OPL2.cpp:49:

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy/Print.h:47:17: error:   overriding 'virtual size_t Print::write(uint8_t)'

  virtual size_t write(uint8_t b) = 0;

                 ^

Mehrere Bibliotheken wurden für "OPL2.h" gefunden
 Benutzt: C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-teensy_midi
 Nicht benutzt: C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-master
Fehler beim Kompilieren für das Board Teensy++ 2.0.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

DhrBaksteen, 2018-07-03:
@saschaamend Hmm surprising that it's complaining about `usb_names.h` missing. It's also not there in the Teensy example code where I copied it from. My suspicion is that the line should've been `#include <usb_names.h>` and I was just lucky with the IDE helping out in resolving the file.

After searching for the file I found it in the Teensy3 folder. It does not look specific to Teensy 3.0 or later, so if changing the include is not helping then maybe including this header in the lib will help? Worst case is that it really is only for 3.0 and later (I'm using a Teensy 3.6) in which case I need to make some more modifications.

@pioneerman2k1 From 
`Mehrere Bibliotheken wurden für "OPL2.h" gefunden
Benutzt: C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-teensy_midi
Nicht benutzt: C:\Users\Pione\Documents\Arduino\libraries\ArduinoOPL2-master`
it looks like you have two versions of the library in the libraries folder. The Arduino IDE does not like that. Try moving out the master and only leave this branch in there. Could that also be the case for you @saschaamend ?

I will push some changes soon that will hopefully fix the usb_names issue and also get rid of a warning. Thanks for the feedback it is very helpful!

pioneerman2k1, 2018-07-05:
i followed the instructions, but still get this error. I also tried it on the Notebook of my wife with a clean install of Arduino IDE. The same error Message..

saschaamend, 2018-07-06:
With the new brench it workt for me. This evening i will test some of my favorite Dos Games..

DhrBaksteen, 2018-07-06:
Excellent!

@pioneerman2k1 I tried to reprodce your issue for a bit, but no 'luck' so far... So maybe the latest commit also fixes your issue automagically

pioneerman2k1, 2018-07-06:
I cant get any sound from it.. The Code work for me, but with teensy ++ 2.0 no sound at all

pioneerman2k1, 2018-07-06:
Same for me.. Wireing is correct but no sound. I ordered an Teensy LC. It is the low cost model but has more capabilities than the ++ 2.0

DhrBaksteen, 2018-07-06:
Looking at the pinout for the Teensy ++2.0 it looks like hardware SPI is not on pins 11 - 13. https://www.pjrc.com/teensy/td_libs_SPI.html 

Can you give the following a try?
Reset = E0
A0 = E1
Latch = C0
Data = B2
Shift = B1

pioneerman2k1, 2018-07-06:
Cant believe it, but it does work for me *happy*

DhrBaksteen, 2018-07-14:
Teensy support has been added to the library and released as v1.4.2
* Added support for MIDI
* Teensy specific examples when SD card is required
* Updated documentation

Closing this issue.

closed 2018-07-14 by DhrBaksteen