DhrBaksteen/ArduinoOPL2 #30: OPL2 pass-through for DosBox

DhrBaksteen, 2018-07-17:
As was mentioned in #29 it would be nice if DosBox can pass its OPL2 data to the OPL2 board to have a real YM3812 play the OPL2 audio. My comments from the PR that is requesting this:

> I'm already working on that :)
> 
> It will require changes to DosBox that I will place in a new repository. The sketches for the OPL2 board I will add here and of course add links from one repo to the other. What I'm aiming for is that it will work with any controller, not just with a Teensy, but that when using a Teenst you can have both OPL2 pass-through and MIDI support.
> 
> I have the pass-through working on one of my test systems which is an Orange Pi that runs a modified version of DosBox, but that is directly communicating to the OPL2 board through the GPIO header. For a more general solution I'm looking into outputting the OPL2 data over a serial line, so it's easy to configure and it will work on any platform.
> 
> Since there is nothing for me to merge here I will close this PR and create an enhancement ticket for people to comment on.

saschaamend, 2018-07-19:
My dreams come true. Cant wait for it. Thank u!!

DhrBaksteen, 2018-07-21:
Modified DosBox code is now available in [ArduinoOPL2.DosBox](https://github.com/DhrBaksteen/ArduinoOPL2.DosBox) repository.

pioneerman2k1, 2018-07-22:
I tried it on my Teensy 2.0 ++ with no Luck. DOSBOX wrote in the Terminal Window, that it is connected ( "OK" ) and then DosBox hangs.
With my Arduino Uno i got no problem.

DhrBaksteen, 2018-07-22:
You did set the USB type of the Teensy to `Serial` right?

pioneerman2k1, 2018-07-23:
Yes, i did.

DhrBaksteen, 2018-08-03:
@pioneerman2k1 Can you check the CPU speed of your Teensy? The lowest I checked so far was 16 MHz on an Arduino Nano. Like you said that works fine. Maybe the baud rate of 115200 is causing issues if the CPU speed is set to someting lower?

I'm new to using the Teensy boards, but could it be that the port that you program it with is different form the serial port for USB serial? I can imagine that when setting the program port in the DosBox config it can connect, but it will crash.

DhrBaksteen, 2018-09-12:
The pass through sketch is now available as of release v1.4.3. Closing this issue.

closed 2018-09-12 by DhrBaksteen