DhrBaksteen/ArduinoOPL2 #31: BBC Microbit SPI support

pappavis, 2018-07-22:
Ter info.  Ik heb jouw Arduino OPL2 code gecompileerd op een Microbit en het werkt!!!
Alleen.. jouw arduino SPI pinouts zijn niet compatible met Microbit. 

It should be a fairly quick fix.

DhrBaksteen, 2018-07-22:
Please keep the discussions on Github in English. That way everyone can be able to understand and participate.

I'm not familiar with this Microbit, but after Googling it, it seems more the fact that its MOSI and SCK signals are on different pin numbers than the Arduino's for example. This is not an issue in the code, but rather something I can add to the wiki on how to connect the OPL2 board.

Nonetheless, cool that it works on yet another platform.

DhrBaksteen, 2018-07-23:
Pinout description added to the wiki https://github.com/DhrBaksteen/ArduinoOPL2/wiki/Connecting

This resolves #31

closed 2018-07-23 by DhrBaksteen