DhrBaksteen/ArduinoOPL2 #36: Bug: setInstrument uses invalid registers

stople, 2018-12-18:
Thanks for a cool project! I am making an OPL synthesizer connected to my digital piano.

SetInstrument uses invalid registers when i == 5. You should change it to something like this (I guess it applies to percussion instruments as well):

		default:	// Melodic instruments...
			for (byte i = 0; i < 11; i ++) {
				byte reg;
				if (i == 5)
					reg = 0xC0 + max(0x00, min(channel, 0x08));
				else
					reg = instrumentBaseRegs[i % 6] + getRegisterOffset(channel, i > 5);

				setRegister(
					reg,
...

DhrBaksteen, 2018-12-21:
Thanks for pointing it out! Register 0xC0 is indeed channel based and not operator based :face_palm:

DhrBaksteen, 2019-01-18:
Issue is fixed by #38

closed 2019-01-18 by DhrBaksteen