DhrBaksteen/ArduinoOPL2 #39: Library causing problems with other pins/existing Arduino sketch

christopherwk210, 2019-01-26:
I have this sketch:

```
#include <SPI.h>
#include <OPL2.h>

#define NUM_ROWS 4
#define NUM_COLS 8

// Row input pins
const int row1Pin = 7;
const int row2Pin = 6;
const int row3Pin = 5;
const int row4Pin = 4;

// 74HC595 pins
const int dataPin = 3;
const int latchPin = 2;
const int clockPin = 12;

boolean keyPressed[NUM_ROWS][NUM_COLS];

// bitmasks for scanning columns
int bits[] =
{ 
  B00000001,
  B00000010,
  B00000100,
  B00001000,
  B00010000,
  B00100000,
  B01000000,
  B10000000
};

OPL2 opl2;

void setup()
{
//  opl2.init();
//  opl2.setMaintainSustain(0, CARRIER, true);
//  opl2.setMultiplier(0, CARRIER, 0x04);
//  opl2.setAttack    (0, CARRIER, 0x0A);
//  opl2.setSustain   (0, CARRIER, 0x04);
  
  for(int colCtr = 0; colCtr < NUM_COLS; ++colCtr)
  {
    for(int rowCtr = 0; rowCtr < NUM_ROWS; ++rowCtr)
    {
      keyPressed[rowCtr][colCtr] = false;
    }
  }

  // setup pins output/input mode
  pinMode(dataPin, OUTPUT);
  pinMode(clockPin, OUTPUT);
  pinMode(latchPin, OUTPUT);

  pinMode(row1Pin, INPUT);
  pinMode(row2Pin, INPUT);
  pinMode(row3Pin, INPUT);
  pinMode(row4Pin, INPUT);

  Serial.begin(9600);
}

void loop()
{
  for (int colCtr = 0; colCtr < NUM_COLS; ++colCtr)
  {
    //scan next column
    scanColumn(colCtr);

    //get row values at this column
    int rowValue[NUM_ROWS];
    rowValue[0] = digitalRead(row1Pin);
    rowValue[1] = digitalRead(row2Pin);
    rowValue[2] = digitalRead(row3Pin);
    rowValue[3] = digitalRead(row4Pin);

    // process keys pressed
    for(int rowCtr=0; rowCtr<NUM_ROWS; ++rowCtr)
    {
      if(rowValue[rowCtr] != 0 && !keyPressed[rowCtr][colCtr])
      {
        keyPressed[rowCtr][colCtr] = true;
        noteOn(rowCtr,colCtr);
//        opl2.playNote(0, 2, 0);
      }
    }

    // process keys released
    for(int rowCtr=0; rowCtr<NUM_ROWS; ++rowCtr)
    {
      if(rowValue[rowCtr] == 0 && keyPressed[rowCtr][colCtr])
      {
        keyPressed[rowCtr][colCtr] = false;
      }
    }
  }
}

void scanColumn(int colNum)
{
  digitalWrite(latchPin, LOW);
  shiftOut(dataPin, clockPin, MSBFIRST, bits[colNum]);
  digitalWrite(latchPin, HIGH);
}

void noteOn(int row, int col)
{
  Serial.println(String(row) + ", " + String(col));
}
```

Which reads a scan matrix hooked up to a keyboard. It works perfectly as expected as long as I don't call `opl2.init()`. If I uncomment that line though, it stops working. Just uncommenting that line alone causes the serial to print lots of output at once when even pressing a single key, and uncommenting everything to do with the OPL2 library causes the serial output to stop altogether.

I'm using all of the default pins for the OPL2 library as stated in the wiki, and the pins I'm using for the scan matrix can be seen in my snippet. If there's something I'm doing wrong please let me know, otherwise there may be an issue with the library.

I'm using an Arduino UNO.

DhrBaksteen, 2019-01-26:
The pins that are used can simply be reassigned using the constructor: `OPL2::OPL2(byte reset, byte address, byte latch)`.

However in your case the problem look to be pin 12. This pin is reserved in the Arduino SPI library for the MISO signal. The OPL2 board is not using this signal, but using it will likely confuse the Arduino. Please try moving it to a different pin.

closed 2019-01-26 by DhrBaksteen

christopherwk210, 2019-01-26:
Sorry, I should have mentioned that I've tried many other pins. Changing out pin 12 for another doesn't seem to help at all. No matter what pins I use, the only thing that causes an issue is initializing the OPL2 library.

christopherwk210, 2019-01-28:
> Hi,
>
> If your issue still exists then I suggest to share a schematic.

I’m actually all settled, I’m not sure what the issue was but I took
everything apart and put it all back together to find it solved. Must have
been a wiring issue on my end, thanks for the help.