DhrBaksteen/ArduinoOPL2 #40: OPL2.play playback issues

michalprochazka02, 2019-02-22:
Hello, I've recently purchased your OPL2 sound board, tested it with my Raspberry Pi and it works great!

I played some of the tunes that were included, which worked fine, but most of the recordings I made didn't sound right, found the issue: in line 127 of file examples_pi/opl2play/opl2play.cpp, replace:

} else {

with

} else if (registerCode < dro.registerMapLength) {

My recordings were made using DOSBox 0.74-2, and after this change they play fine.

DhrBaksteen, 2019-02-22:
Hi,

Thanks for reporting this! I'll add this fix to the upcoming release.

Out of interest, what games did you record from? I think that the issue you found has something to do with DosBox using a dual OPL2.

michalprochazka02, 2019-02-22:
I recorded some tunes from Ultima 6, Duke Nukem 3D, Tyrian and The Secret of Monkey Island. All of these games were set to output to AdLib. I included my recordings in a zip file:

ultima.dro - 1 of the channels has a wrong waveform, out of tune
ultima2.dro - high pitched tone at start, then some channels have a wrong waveform
tyrian.dro - Some channels have a wrong waveform
duke3d.dro - During playback, it sometimes changes tempo/skips a few notes, some channels are missing
monkey.dro - The tempo is all over the place, some channels have a bad waveform

The tempo issues are still present after the fix, but that may be because I'm using a Raspberry Pi 1, which is WAY slower than the Pi 2 or 3

[tunes.zip](https://github.com/DhrBaksteen/ArduinoOPL2/files/2894923/tunes.zip)

DhrBaksteen, 2019-02-23:
Thanks for sharing your captures. It turns out that DosBox is apparently clearing registers on a second OPL2 that is never used.

I've found some other issues as well in both the Raspberry Pi and the Arduino sketch regarding timing. The Tyrian tune was unplayable for example.

michalprochazka02, 2019-03-13:
Regarding the timing, I have tried another tune and it skipped horribly at the start. The cause is this: 

"The iShortDelayCode value must have one added to it, as per v1.0 (so a short delay of 2 == a 3ms delay)
The iLongDelayCode value must have one added to it, then be multiplied by 256 (or "<< 8"), so a long delay of 2 == a 768ms delay"
(source: http://www.shikadi.net/moddingwiki/DRO_Format)

But your code contains (@ lines 124 & 126 of opl2play.cpp):

"delay(value);"
and
"delay(value << 8);"

Changing them to:

"delay(value + 1);"
and
"delay((value + 1) << 8);"

fixes the issue for me.

DhrBaksteen, 2019-03-14:
You are right, I missed this on the Pi code. Reopening the issue

commit be86f970 (2019-02-25): Fixes to v1.4.4

The following fixes are made bringing the library to v1.4.5

* Some cleanup of the code in OPL2.cpp
* Fixed #40 where OPL2Play did not take into account that registerMap
entries > 127 should be ignored
* Fixes to PlayDRO and PlayIMF examples for Arduino and Teensy where for
some tunes updating of the timer was ignored causing all delays to be
skipped
* Updating the library to v1.4.5
PR #41 (2019-02-25): Fixes for 1.4.4
Fixes #40
commit d016fcc1 (2019-02-25): Fixes to v1.4.4

The following fixes are made bringing the library to v1.4.5

* Some cleanup of the code in OPL2.cpp
* Fixed #40 where OPL2Play did not take into account that registerMap
entries > 127 should be ignored
* Fixes to PlayDRO and PlayIMF examples for Arduino and Teensy where for
some tunes updating of the timer was ignored causing all delays to be
skipped
PR #42 (2019-03-17): OPL2Play DRO fix
This fixes a bug in OPL2Play where DRO file delays were 1ms too short,
causing some delays to be skipped completely and songs to be broken.

This fixes #40
closed 2019-03-17 by DhrBaksteen