DhrBaksteen/ArduinoOPL2 #48: Bug when setting registers for melodic instruments

virtuaCode, 2019-09-30:
https://github.com/DhrBaksteen/ArduinoOPL2/blob/61e03ff184a575fb7db65fb2ffecdecd9f1b98fe/src/OPL2.cpp#L260-L266

The parameters for the registers C0...C8 are stored in the instrument data at `instrument[5]`. Since we always do `i + 1` when accessing the instrument data, the condition at line 260 have to be `i == 4`. Also accessing the `instrumentBaseRegs` array has to be done this way:

```reg = instrumentBaseRegs[i == 5 ? 4 : i % 6] + getRegisterOffset(channel, i > 5);```

After I fixed this issue I played some notes with the instrument in `midi_instrument.h` and realized that for every patch the "synthtype" bit is inverted. :open_mouth: Can you confirm this?

DhrBaksteen, 2019-10-01:
The library code is correct. The problem is with the export of the new instruments. The OPL2 registers in the instruments are not supposed to be in order, but 0x20, 0x40, 0x60, 0x80, **0xE0, 0xC0**, etc. The original `instruments.h` has this correct, but the midi exports have registers 0xC0 and 0xE0 switched.

DhrBaksteen, 2019-10-02:
I'm in doubt whether there is actually anything wrong with the instruments. Trying to swap the two columns results in very broken instruments. My suspicion is that there is additional bits left on the 0xE0 registers that were not filtered out during export. I need to check this with the BNK file where the instruments came from...

Still comments on the instruments files are wrong. Also length of `instrumentBaseRegs` in opl2.h should be declared as 6.

closed 2019-11-10 by DhrBaksteen