DhrBaksteen/ArduinoOPL2 #59: OPL2 via Serial on real computers.

josephillips85, 2020-07-26:
Hi Maarten

im wondering if this is posible like the OPL2LPT project but with serial.

josephillips85, 2020-07-26:
I tried on virtualbox and mapped the serial port using adtracker changing the port to serial. i see that arduino is receiving data from the computer but no sound.
i tried few things like chaging baud rate initialize in DOSBOX and switching to Virtualbox

Is requiered a some initialization before getting the stream?

DhrBaksteen, 2020-07-27:
I'm not very familiar with how serial works on these old systems, but if sending a byte serial is a blocking operation then you could be looking at quite a significant performance impact on these old systems. You also need a device like a MAX232 since the PC is using different voltages. Let alone a TSR program for DOS that hijacks the port. So possible... yes, but using the parallel part like OPL2LPT does is a better solution for these old systems.

I'm not sure what Adtracker (Adlib Tracker II?) is sending, I've not tried it. What the serial passthrough sketch is expecting is a very simple pair of address + data bytes at 115200 baud. There is no additional setup.

josephillips85, 2020-07-27:
Well I'm tried on virtualbox but doesn't work what I will tried on a old laptop that have rs232 and see the response on a real hardware.

So should work if adtracker send the data of adlib to the serial port right ?

Thanks

DhrBaksteen, 2020-07-28:
> So should work if adtracker send the data of adlib to the serial port right ?

I can't guarantee you that. I don't know how / what this program sends because I've not used it. Like I said, if it's sending byte pairs of address + data @ 115200 baud then it is likely to work with the SerialPassthrough sketch.

josephillips85, 2020-08-17:
Hey

i guess i have a advance on this i already patch a game to send data to 3f8 (serial port) and the arduino is getting the data
also the flash pattern (RX) on arduino look equal that when im running via dosbox.

However for one reason im not sure yet if is a issue with baudrate configuration or something (games don't allow set serial baudrate parity etc) because they are looking for adlib port but the register and value are not consistent. sometimes play one or two notes on adlib.

if you have any idea of what else to check let me know 👍

josephillips85, 2020-08-24:
Success 

A proof of concept is working.

Thanks for your recommendation 

https://twitter.com/josephillips/status/1297725755070328832

DhrBaksteen, 2020-08-26:
Good to hear that!

I will close this ticket then.

closed 2020-08-26 by DhrBaksteen