DhrBaksteen/ArduinoOPL2 #71: Line out sounds good, speaker out has lots of noise

isnotinvain, 2020-12-13:
Hello!

Been having fun with my opl3 Duo. The line out sounds pretty good, even when amplified by powered speakers or my audio interface up to audible levels. But the speaker out has a lot of buzzing / humming noise even when not playing anything, even with the volume knob turned all the way down.

Is that expected? If not, any ideas what to check for? 

Thanks!

DhrBaksteen, 2020-12-13:
Hi,

Some background hiss is to be expected from the amplifier. How are you powering the board? If you can provide a recording of the noise maybe it points to an issue.

isnotinvain, 2020-12-13:
I'm powering the board via the 3V3 pin of my raspberry pi (just following the wiring instructions). The raspberry pi is powered by the official wall cable.

I'll try and make a recording -- the noise is a white noise sound, and it's completely independent of the volume knob, it's just present at the same volume at all times. Actually, when I plug in speakers the noise is a lot less noticeable, but I've tried a few different headphones and they really pick up the white noise.

My end goal is to play the board like a midi synthesizer and use headphones to listen, so I'd like to figure out how to get a little less noise if that's possible. Thanks!

DhrBaksteen, 2020-12-14:
The way you describe it the noise is "normal". The TDA2822 is only a very simple stereo amplifier, but you have a few options to reduce the noise when you're using headphones:

1. By bridging C33 and C34 you will reduce the noise. (I did not test this well enough to be sure about distortion in all scenarios)
2. You can also remove C33 and C34 and connect the legs that go to C30 / C32 directly to ground. According to my dev notes it gets rid of the noise and lowers the volume, but also introduces distortion on the mid range of the volume. Playing around with the values for R14 and R15 may help. But for this also I didn't really investigate this because making this connection to GND seemed strange.

isnotinvain, 2020-12-14:
OK thanks that makes sense. Does that mean that the line out should have a cleaner signal? If that's the case do you think if I bought a fancier amplifier chip of some kind I could hook that up to the line out and get better results? Any recommendations on nicer amplifiers? I'd definitely be willing to get an additional board in the $10 range for this project if you think that'd get better sound quality.

Thanks again for your help!

DhrBaksteen, 2020-12-15:
The line out comes directly from the chips and is a much cleaner signal. You would indeed get good results by connecting your own amp to the line output. Most headphone amplifiers will do. I'd give the PAM8403 a try, it seems like a good low-noise stereo amplifier and modules are quite cheap.

isnotinvain, 2020-12-15:
OK thanks, I'll try that!

lethal-guitar, 2021-01-12:
@DhrBaksteen I'm also interested in using the line out. I got the pre-assembled OPL2 board. AFAICT, it only has the amplified 3.5mm jack. Is the line out only for the OPL3 board? Or can I connect some cables directly to one of the ICs for getting the line out?

DhrBaksteen, 2021-01-12:
@lethal-guitar, the OPL2 board does not have a dedicated line output like the OPL3 Duo board. You can grab unamplified audio signal from pin 1 of the LM358 (IC3) or from C9 at the pin going to the volume potentiometer.

isnotinvain, 2021-01-13:
Just an update -- I plugged the line-out into my nice audio interface and the signal there is good.

The noise in the speaker-out is actually really clearly influenced by what the raspberry pi is up to, you can hear it change when the raspberry pi is busy, or even when you move the mouse around or launch a program. Would that mean the noise has to do with the power signal maybe? I'm very much a beginner when it comes to hardware, so I have no idea if that makes sense.

DhrBaksteen, 2021-01-13:
These noises often come from the power supply indeed. Especially when powered over USB you can have issues with cross-talk being picked up and amplified. Other causes of unwanted noise can be long hookup wires or simply the traces from a breadboard can induce unwanted noise. 

Sometimes simply adding more or bigger bypass capacitors will work, you could also think of an LC-filter built from an inductor and capacitor, or you could take it to the extreme using a separate power supply, filters and using optocouplers to transfer signals 🙂. I'm not sure about ferrite beads around cables to get rid of unwanted noise, but some say it helps too.

Your nice audio interface probably has a lot of filtering and good grounding to get rid of all the unwanted noise.

isnotinvain, 2021-01-15:
Thanks for all the details! I'm still hooked up to a breadboard as well, I should see if it gets better w/o that, that's an easy thing to try.

closed 2021-09-20 by DhrBaksteen