DhrBaksteen/ArduinoOPL2 #73: min / max macros in opl2.h cause issues when using opl2.h with other libraries

isnotinvain, 2020-12-20:
Filing this so I don't forget -- I'll try and send a PR with a fix soon.

I'm not a C++ expert but I did some googling after hitting some compiler errors when I tried to use `std::min` in a file that includes `OPL2.h` and found that defining min/max macros is discouraged (for this reason).

This workaround works:
```
#include <OPL2.h>
#undef min
#undef max

... rest of your code... 
```

I think probably `std::min` and `std::max` could be used instead, at least for raspberry pi. Or this macro could move to the relevant cpp files instead.

DhrBaksteen, 2021-01-17:
In the library and examples these functions are only used to clamp values to a valid range. Using `std::` is not an option for Arduino due to its memory limitations. The upcoming library update contains a helper fucntion `clamp` to take care of this and that's also where the Arduino / Raspberry Pi / others distinction will be made to prevent spreading complier directives throughout the code in order to keep it simple and readable.

isnotinvain, 2021-01-17:
ok sounds good, thanks!

DhrBaksteen, 2021-01-21:
Please reopen if the issue persists

closed 2021-01-21 by DhrBaksteen

isnotinvain, 2021-01-26:
Great, thanks! I'll back out my `undefine`s and see how it goes

PR #77 (2021-01-21): Clamping function
This replaces the `min` and `max` macros with a new function `clampValue` to fix #73.

Tested with
- Arduino
- Raspberry Pi
- Teensy
- ESP8266 (NodeMCU)
- STM32 (Blue Pill)