DhrBaksteen/ArduinoOPL2 #74: OPL3 class seems to lack OPL3-specific addressing

jonkerj, 2021-01-18:
I previously wrote some code to do MIDI-to-OPL2 and play old tunes through a MIDI console. I recently upgraded to (single) OPL3 using the OPL3Duo, which sounded great as a drop-in replacement. To level up my game, I enabled 4-op mode, but I got a feeling I could not achieve full 6-voice polyphony: notes sounded like they were getting "stolen" by my voice allocator.

I did some debugging, and the voice allocator works fine. Actually, it did before on 6-voice 2-op OPL2, so I dug a little deeper. My next guess was there was something wrong with the BANK/A1 pin, which would effectively halve the amount of voices. And also, it would prevent the chip from going into OPL3 mode. I started comparing the sounds to an emulated OPL3, and it could be the case.

Looking at the code:
* OPL3.cpp does not have its own `setChipRegister`, so defaults to the OPL2 implementation, which masks the register with 0xff. Registers 0x100 and up can never be addressed in this way
* OPL3.cpp does have a `setChannelRegister` and `setOperatorRegister`, but BANK/A1 only get driven when `(channel >> 8) & 0x01`, which is of course never the case

I am trying to get my head around the datasheet/programming guide to come up with a PR to address these issues, but if anyone is more fluent in YM262, I would appreciate some help :-)

DhrBaksteen, 2021-01-18:
I think you're using the OPL3 class directly?  When I wrote OPL3.cpp I intended it as a base for the OPL3Duo class, so I have not fully tested it stand alone.

1. The `setChipRegister` for the OPL3 class was indeed missing and I have staged it locally for the upcoming update as
```c++
void OPL3::setChipRegister(short reg, byte value) {
	chipRegisters[getChipRegisterOffset(reg)] = value;
	write(reg >> 8, reg & 0xFF, value);
}
```

2. For the `setChannelRegister` and `setOperatorRegister` of the OPL3 class I see I completely messed up here cause indeed it should not be `(channel >> 8) & 0x01`! It can be something like `byte bank = (channel / CHANNELS_PER_BANK);` as in the OPL3Duo class.

jonkerj, 2021-01-19:
Yes indeed. I am using OPL3 directly. I will try the suggestions, was already peeking at the OPL3Duo code, where they seem more appropriate.

jonkerj, 2021-01-19:
Yes, that works. I'll PR it for good measure.

PR #75 (2021-01-20): Fix OPL3
It seems the OPL3-specific code is not fully able to use the OPL3 features at all. This PR fixes this. Tested on Arduino with OPL3.

Fixes #74
commit 9a18e581 (2021-01-20): Fix OPL3

Fixes #74
closed 2021-01-20 by DhrBaksteen