DhrBaksteen/ArduinoOPL2 #82: OPL2 operates on a 4 MHz resonator instead of 3.59 MHz.

Opekusha, 2021-01-27:
How to programmatically adjust the frequency? How to get into unison? Synthesizers usually have this tuning. I haven't figured out which library function to use yet.

DhrBaksteen, 2021-01-27:
The library is built around using a 3.579 MHz clock for the YM3812, as you normally would for this chip. So there is no function to adjust for different clock frequencies. Using a different clock frequency means that you have to recalculate all the F-numbers and possibly also use different frequency blocks to be in tune with a normal 3.579 MHz clock.

Nb. `setFrequency` also assumes a 3.579 MHz clock for its block and F-number calculation.

Opekusha, 2021-01-28:
Thanks I got it. It's easier to find a resonator. The question has been settled.

closed 2021-01-28 by DhrBaksteen