DhrBaksteen/ArduinoOPL2 #84: Just a question

VectorFever, 2021-06-05:
How good is UART speed in SerialPassthrough for OPL2? Is everything running without lags? I have ordered parts i need to make my usb AdLib and I'm going to use it with DOS trackers,midi and vgm players. How good it in very dynamical music?

DhrBaksteen, 2021-06-05:
The speed is good. With gaming and MIDI playback there should be no lag. With very complex Adlib Tracker songs you may experience some lag for example when a song makes very heavy use of macros.

VectorFever, 2021-06-06:
Ok. Does increasing UART speed make any sense? I think for that dosbox x need to be edited.

DhrBaksteen, 2021-06-06:
No really, the baud rate is currently set to 115200 wich is more than enough bandwidth. But the logistics around the serial transfer can come into play here. For example how is the serial buffer transferred? And can this cause delays in the emulator? Could there be performance issues due to using the regulat Dosbox Adlib frontend to send the data?

VectorFever, 2021-06-06:
Ok. Thank you! I think it will be enough for standard DOS gaming!

closed 2021-06-15 by VectorFever