DhrBaksteen/ArduinoOPL2 #85: A lot of noise in headphones.

VectorFever, 2021-06-24:
Hi! I made OPL2 using your schematic and it seems to be working perfect. But when headphones connected there is interference. I hear *shhhhhhhh* always and 50 hz buzz when my hands near the board. Also i can hear pc processes like scratching sound when move cursor. When using speakers there is no such problem. Is something with the power? Maybe ams 1117 in my Chinese Arduino is not enough for such current and it causes voltage drops followed by interference
[image]
?

VectorFever, 2021-06-24:
Maybe it's caused by 0.1μF capacitor between lm386 negative and ground. On every lm386 schematic(lm386 headphone preamp schematics too) it's connected straight to ground. So when there is no AC coming to capacitor it acts like air and like mini antenna.

prosper00, 2021-07-07:
LM386 is notorious for noise; there are lots of various guides and tips online for cleaning up its output. It also sounds like you're running it from USB power from your PC. USB provides really noisy power. Try running it from its own power source. Ideally, 6V or so, to give your AMS1117 enough headroom to regulate it down to 5V and clean out some of the noise.

VectorFever, 2021-07-11:
Here is interesting part... I have that noise only in headphones. In speakers there is crystal audio.

VectorFever, 2021-07-11:
I have chosen that board factor because i made it as USB AdLib sound card to use with DosBox. USB Type-C on input, 3/5mm jack on output. Small USB AdLib :)

prosper00, 2021-07-11:
Could still be a ground problem: headphones are touching you,  and therefore you are capacitvely coupled into the circuit.  Speakers arent in contact with you. If your grounds arent  carefully considered,  this can certainly cause mains hum

closed 2021-09-20 by DhrBaksteen