DhrBaksteen/ArduinoOPL2 #88: Piano board support.

VectorFever, 2021-07-11:
Is it possible to increase number of keys? As it uses 74HC165 it can be connected to others many times. I have few OPL2s and now i want to make keyboard :) . But having only one octave is problem. Im from Russia,sorry for my broken english.

DhrBaksteen, 2021-07-11:
Hi,
This is possible for sure. The example is meant to be used with the Piano Board that only has a single octave, but (to keep my answer as simple as possible) let's say for example that you had 3 Piano Boards, one for each octave. 

Each piano board will have a different pin for /SS (select), all other pins are connected together. We will create 3 PianoKeys instances, one for each octave and with a different pin for /SS (pins 5, 6, and 7 in this case).
``` C++
PianoKeys keysOctave[3] = {(5), (6), (7)};
```

Now in the main loop of the example we need to wrap all code in a `for` loop to check each octave for key presses. We no longer need the code that increments and decrements the octave because it now depends on the piano board that registered the key press. Lastly we need to register what key was pressed per OPL2 channel by taking into account the octave + key.

``` C++
void loop() {
    for (byte o = 0; o < 3; o ++) {
        octave = 3 + o;        // Our actave will depend on the Piano Board that regitsered the key press with a base octave of 3.
        keysOctave[o].updateKeys();        // Update keystates for the current octave.

	for (int i = KEY_C; i <= KEY_C2; i ++) {
                // Did we register a key press on the current octave?
		if (keysOctave[o].wasKeyPressed(i)) {
			int opl2Channel = getFreeChannel();
			if (opl2Channel > -1) {
				keyChannel[opl2Channel] = o * 12 + i ;     // Take into account octave + key!
				if (i == KEY_C2) {
					opl2.playNote(opl2Channel, octave + 1, NOTE_C);
				} else {
					opl2.playNote(opl2Channel, octave, i);
				}
			}
		}

                // Did we register a key release on the current octave?
		if (piano.wasKeyReleased(i)) {
			int opl2Channel = getKeyChannel(o * 12 + i);      // Look for the octave + key
			if (opl2Channel > -1) {
				keyChannel[opl2Channel] = -1;
				opl2.setKeyOn(opl2Channel, false);   
			}
		}
	}
    }

    delay(20);
}
```

You can of course custom build your keyboard where you string more 74165s together and integrate the PianoKeys code with this simple example. You would have to keep the key states in a byte array that can hold the state of each 74165 instead of a single `int` like the PianoKeys code does now. Then by iterating the array and shifting the bits you can find what keys were pressed and play the correct notes.

VectorFever, 2021-07-11:
Wow! Huge thanks! I will test it in several weeks and if everything will be ok i will assemble my keyboard! I'm going to use 6 octaves. But it will be not soon.

VectorFever, 2021-07-11:
Is shift registers pulled up or down with resistor networks?

DhrBaksteen, 2021-07-11:
On the Piano Board the resistor networks function as pull ups

closed 2021-09-20 by DhrBaksteen