DhrBaksteen/ArduinoOPL2 #90: OPL3 Duo with OPL3 demotune hangs with Raspberry Pi

sneakernets, 2021-08-25:
Hi, I'm using an original Raspberry Pi B model, and the OPL3 version of the demotune program hangs about 5 seconds in, playing the same note forever. The other demotune program doesn't have this issue.

The hardware test seems fine, but I don't know what that program is really supposed to sound like.

DhrBaksteen, 2021-09-20:
I've not been able to reproduce this issue. If it still exists then please provide more info like what software versions are running on your Pi.

sneakernets, 2021-09-20:
I'm using the latest Raspberry Pi OS release on a Pi 1 B model, that much I know. I will make a video to demonstrate where the demo song is hanging up. for now I can put an arrow where the song is hanging on voice1:
```C
                                                         v   
const char voice1[] = "i96t150o5l8egredgrdcerc<b>er<ba>a<a>agdefefedr4.regredgrdcerc<b>er<ba>a<a>agdedcr4.c<g>cea>cr<ag>cr<gfarfearedgrdcfrc<bagab>cdfegredgrdcerc<b>er<ba>a<a>agdedcr4.cro3c2r2";
const char voice2[] = "i96o4l8crer<br>dr<ar>cr<grbrfr>cr<grbr>crer<gb>dgcrer<br>dr<ar>cr<grbrfr>cr<grbr>ceger4.rfrafergedrfdcrec<br>d<bar>c<agrgd<gr4.o4crer<br>dr<ar>cr<grbrfr>cr<grbr>cege";
const char voice3[] = "i2o3l8r4gr4.gr4.er4.err4fr4.gr4.gr4.grr4gr4.er4.er4.frr4gr4>ccr4ccr4<aarraar4ggr4ffr4.ro4gab>dr4.r<gr4.gr4.err4er4.fr4.g";
```

DhrBaksteen, 2021-10-11:
Can you check what version of Wiring Pi is running on both Pis with ` gpio -v` ? Maybe there is a difference there, or could there be some other software running in the background interfering with the gpio?