DhrBaksteen/ArduinoOPL2 #97: Reset-Function when using Serial Passthrough

pioneerman2k1, 2025-01-14:
Is it possible to implement a resetfunction in serial mode when using an arduino? 
Maybe with specific register and data values.

DhrBaksteen, 2025-01-14:
Hi,

Yes that's a simple change to let it reset the OPL when you write for example 255 to register 0.
I'll have a look at it this week.

DhrBaksteen, 2025-01-19:
Writing value 165 to register 255 (and bank 0 on OPL3 Duo!) will now do a hard reset of the OPL2 or OPL3.

This specific register and data value was chosen to make it very unlikely that an emulator or software will reset the chips unintentionally which may cause audio to break.

After doing a hard reset using the serial passthrough there should be a delay of at least 2ms before sending additional data to prevent potential flooding of Arduino's serial buffer.

closed 2025-01-19 by DhrBaksteen