DhrBaksteen/ArduinoOPL2 #99: Slowdowns in SerialPassthrough

pioneerman2k1, 2025-01-19:
Hi Marteen,

i am not sure it is because of my choice to use c# for programming or i messed something up. But in "simple" VGZ Files like Funky from Jill of the Jungle everything is fine with timing. But when playing the Intro from Monkey Island 1 the first seconds playing fine and when the rithm kicks in the are slowdowns. Any suggestions? Maybe i am calculating the breaks false? I am at a complete loss...

My Code is very simple:

        **private async Task PlayVGMAsync(byte[] vgmData)
        {
            const int SAMPLE_RATE = 48000; // 48.0 kHz

            using (MemoryStream stream = new MemoryStream(vgmData))
            using (BinaryReader reader = new BinaryReader(stream))
            {
                // 2. Lese relevante Header-Felder
                reader.BaseStream.Seek(0x04, SeekOrigin.Begin);
                uint eofOffset = reader.ReadUInt32(); // Dateigröße
                reader.BaseStream.Seek(0x08, SeekOrigin.Begin);
                uint version = reader.ReadUInt32(); // VGM-Version
                reader.BaseStream.Seek(0x0C, SeekOrigin.Begin);
                uint gd3Offset = reader.ReadUInt32(); // GD3-Tags
                reader.BaseStream.Seek(0x18, SeekOrigin.Begin);
                Program.YM3812_CLOCK = reader.ReadUInt32(); // OPL2-Taktfrequenz
                reader.BaseStream.Seek(0x34, SeekOrigin.Begin);
                uint dataOffset = reader.ReadUInt32(); // Daten-Offset

                textBox1.AppendText(($"Titel: {gd3Offset} ") + "\r\n");
                textBox1.AppendText(($"Dateigröße: {eofOffset} Bytes") + "\r\n");
                textBox1.AppendText(($"Version: {version:X}") + "\r\n");
                textBox1.AppendText(($"YM3812-Takt: {Program.YM3812_CLOCK} Hz") + "\r\n");
                textBox1.AppendText(($"Daten-Offset: {dataOffset} Bytes") + "\r\n");
                textBox1.ScrollToCaret();

                // Überspringe den VGM-Header (64 Bytes)
                stream.Seek(dataOffset, SeekOrigin.Begin);

                while (stream.Position < stream.Length)
                {
                    byte command = reader.ReadByte();

                    switch (command)
                    {
                        case 0x5A: // OPL2-Befehl
                            byte register = reader.ReadByte();
                            byte data = reader.ReadByte();
                            await SendToOPL2Async(register, data);
                            break;

                        case 0x61: // Pause für N Samples
                            ushort waitSamples = reader.ReadUInt16();
                            int delayMs = (int)(waitSamples / (SAMPLE_RATE / 1000.0));
                            textBox1.AppendText(($"Warte {delayMs} ms") + "\r\n");
                            await Task.Delay(delayMs); // Asynchrone Pause
                            break;

                        case 0x62: // 1/60 Sekunde warten
                            textBox1.AppendText(($"Warte {17} ms") + "\r\n");
                            await Task.Delay(17); // ≈16.67 ms
                            break;

                        case 0x63: // 1/50 Sekunde warten
                            textBox1.AppendText(($"Warte {20} ms") + "\r\n");
                            await Task.Delay(20); // 20 ms
                            break;

                        case 0x66: // Ende der VGM-Daten
                            return;

                        case >= 0x70 and <= 0x7F: // 0-15 sample delay
                            waitSamples = (ushort)(command - 0x70 + 1);
                            await Task.Delay((int)(waitSamples / (SAMPLE_RATE / 1000.0))); // Umrechnung in Millisekunden
                            break;

                        default:
                            // Unbekannte Befehle überspringen
                            //byte registerUnbekannt = reader.ReadByte();
                            //byte dataUnbekannt = reader.ReadByte();
                            //await SendToOPL2Async(registerUnbekannt, dataUnbekannt);
                            textBox1.AppendText(($"Unbekannter Befehl: {command:X2}") + "\r\n");
                            break;
                    }
                }
            }
        }

        private async Task SendToOPL2Async(byte register, byte data)
        {
            byte[] command = { register, data };
            await Program.serialPort.BaseStream.WriteAsync(command, 0, command.Length);
        }**

DhrBaksteen, 2025-01-19:
Hi,

The problem is probably with how the delays are being handled. The VGM file assumes that all data is sent to the OPL2 instantly, but that's not what happens in reality. It takes some time to send the serial data and I see that you're `awaiting` each serial write. On simple songs like those of Jill of the Jungle there isn't much data to send, so the time it takes can be ignored, but once a song gets more complicated there is much more data to send and the time to send all this data builds up. If you add the delays from the VGM on top of this then playback speed becomes very irregular.

So some things to take into account:
* Build up a buffer of all the data that you need to send to the OPL2 in between delays
* Once you hit a delay command in the VGM file you send the buffer to the OPL2. Preferably your code is not blocking (or awaiting) when sending the data to the serial port
  * If you must await sending the serial data then before sending you grab a timestamp (preferably in microseconds)
  * Once all data is sent you calculate the difference between the current time and your timestamp to know how long it took to send the data
  * Subtract this time from the VGM's delay time to compensate for the time it took to send all data in the buffer (in case you get a negative delay then don't delay but immediately process the next frame)
  * Keeping the delays in microseconds is preferable

Hopefully this helps you!

pioneerman2k1, 2025-01-19:
Amazing!
when you read it like that, you immediately think: yes, of course! If you're so set in your ways of thinking, then you don't think about something like a buffer. That's exactly what it will be. I'll do a bit more coding next weekend.

Thank you!

closed 2025-01-28 by pioneerman2k1